JimBrown/MarlinTarantula #5: G29 Auto Bed Level puts nozzel too far above bed

gregspruce, 2017-06-13:
### Expected behaviour
Expect G29 to locate z0 ~1 sheet of paper above bed.

### Actual behaviour
After running G28 to home all axes, nozzle is 1 sheet of paper above bed (manually leveled bed on install).
Then I run G29, after hitting all of the points, G0 Z0 leaves the nozzle slightly above the bed.

I set my z_probe_offset_from_nozzle by turning off software endstops, lowering extruder until the probe activated (M119). Then I raised the Z by .1mm increments until the probe reported OPEN. This gave me an offset of -3.1

Like I said, G28 sets Z0 to expected height, but after autobed leveling (using bilinear), Z0 is above the expected location.

### Steps to reproduce the behaviour
Above, config files attached.
Tevo Tarantula with e3d V6 hotend, Pretty sure I cloned MarlinFirmware:bugfix-1.1.x on 6/7/2017.

#### please add your Configuration.h and Configuration_adv.h to a zip file and attach it to this issue
[G29Issue_ConfigFiles.zip](https://github.com/JimBrown/MarlinTarantula/files/1071353/G29Issue_ConfigFiles.zip)

JimBrown, 2017-06-13:
The Z Offset can change slightly from day to day, or even change with what type of material is being used, therefore I leave Z_PROBE_OFFSET_FROM_EXTRUDER set to 0 (zero.)

Instead, I add M851 Zn.nn to the beginning of my start script which allows me to make fine adjustments whenever I need to without having to flash the firmware.

To find that initial offset, I have an STL that simply prints a one layer 30 mm x 30 mm square. I eyeball a value for M851 and try it. If it's not right, I adjust the value slightly and try again. Repeat until satisfied.

I know this does not answer your issue, but I am not one of the programmers of the firmware, I really only know the configuration files. This is how I do it. Hope that helps.

If you really want to get to the bottom of your issue, I suggest posting it in the Marlin Firmware repository.

Regards,
...jim

closed 2017-06-13 by JimBrown

giorgalis, 2018-04-12:
Hello, i am having exact the same issue. Have you found any solution on this?