JimBrown/MarlinTarantula #7: Compile with UBL returns undefined reference error

Kraligor, 2017-08-07:
### Compile Error
When I compile with UBL activated, I get an error:
```
sketch\G26_Mesh_Validation_Tool.cpp.o: In function `unified_bed_leveling::G26()':

sketch/G26_Mesh_Validation_Tool.cpp:238: undefined reference to `sync_plan_position_e()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```
Arduino 1.6.8/Mega 2560/Marlin 1.1.14

Any ideas? I've seen it reported in the Marlin repo, but it doesn't seem to affect everbody.
Cheers, Oli

[Marlin.zip](https://github.com/JimBrown/MarlinTarantula/files/1206443/Marlin.zip)

roygilby, 2017-08-08:
I tried changing my copy of Marlin_Tarantula to UBL this morning, and got this said same compile error.  I understand from my reading on this bug, that it is not being fixed in Marlin Version 1.1.#, and instead is scheduled to be fixed for release in the new Version 2.# which is due soon.  To allow the  developers to concentrate on Version 2.# development, Version 1.1.# has been frozen.

To get your Auto Bed Levelling working in the meantime, I'd suggest trying "AUTO_BED_LEVELING_BILINEAR" instead.  The code for Bilinear Bed Leveling is a bit more mature, and as it uses different code modules, you should avoid whichever code module is generating your problematic error.   Bilinear is the next best thing to Unified, and works reliably for me. 

Hope that helps.

Roy Gilby

JimBrown, 2017-08-08:
I use UBL and am not seeing this. A couple of things though, I'm using Arduino IDE 1.8.3. My understanding is that 1.6.8 is no longer supported with 1.1.x and bugfix-1.1.x. Second, the latest bugfix-1.1.x has turned off G26 by default. I did turn it on and try a verify in Arduino IDE with no errors. Turned it back off since I don't need it (saves program space.)

You can download a new copy of my branch, or you can edit configuration.h, find `#define UBL_G26_MESH_VALIDATION` and comment it out.

@roygilby, Can you post a link to the discussion of this?

Note: my "EasyConfig" branch is a copy of the bugfix-1.1.x branch.

Also note: I update my branches from the Marlin repository almost every day.

Thanks,
...jim

Kraligor, 2017-08-09:
Thanks for your input both!
It works now that I updated to latest Arduino version & bugfix branch. I had it in the back of my mind that it required 1.6.8 to compile successfully.. never thought of trying another version.

Looking forward to returning home from vacation and trying the firmware!

Cheers
Oli

JimBrown, 2017-08-09:
Arduino 1.8.3 did spit out warnings when compiling the old Marlin 1.0, hence the need for 1.6.8 with that.

closed 2017-08-09 by Kraligor