JimBrown/MarlinTarantula #8: Config with BLTOUCH does not compile

luudee, 2017-08-09:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Compile Error
I use the Configure from the examples directory for Tevo Tarantula
When I compile with `BLTOUCH` defined, I get an error:

In file included from sketch/MarlinConfig.h:39:0,
                 from /home/rudi/marlin/MarlinTarantula-EasyConfig/Marlin/Marlin.ino:31:
SanityCheck.h:587: error: #error "You must enable either Z_MIN_PROBE_ENDSTOP or Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use a probe."
     #error "You must enable either Z_MIN_PROBE_ENDSTOP or Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use a probe."
      ^
SanityCheck.h:619: error: #error "Z_MIN_PROBE_REPEATABILITY_TEST requires a probe: FIX_MOUNTED_PROBE, BLTOUCH, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."
   #error "Z_MIN_PROBE_REPEATABILITY_TEST requires a probe: FIX_MOUNTED_PROBE, BLTOUCH, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, or Z Servo."
    ^
Multiple libraries were found for "LiquidCrystal.h"
 Used: /home/rudi/Arduino/libraries/LiquidCrystal
 Not used: /tools/arduino-1.8.3/libraries/LiquidCrystal
exit status 1
#error "You must enable either Z_MIN_PROBE_ENDSTOP or Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN to use a probe."

[Configuration.h.zip](https://github.com/JimBrown/MarlinTarantula/files/1212304/Configuration.h.zip)
[Configuration_adv.h.zip](https://github.com/JimBrown/MarlinTarantula/files/1212305/Configuration_adv.h.zip)

luudee, 2017-08-09:
Aww,  never mind, I just saw it ... comment out "#define MANUAL" for it to work ... dear jesus ... I searched for BLTOUCH but did not look five line below ... :)

Please delete the above issue ...

JimBrown, 2017-08-09:
Glad you found it. I've had those "what the heck" moments too. ;-)

I'll see if I can put something in there to make it clearer.

Regards,
...jim

closed 2017-08-09 by JimBrown

luudee, 2017-08-11:
On 08/10/2017 04:39 AM, Jim Brown wrote:
>
> Glad you found it. I've had those "what the heck" moments too. ;-)
>
> I'll see if I can put something in there to make it clearer.
>
> Regards,
> ...jim
>

Hi Jim,

if you add some warning, may be also add that after commenting
out MANUAL, you MUST enable one of the others or you will end
up without bed leveling, lol ...

Learning the hard way is all the fun ;)

Thanks for all the hard work !

Best Regards,
rudi

-- 
Rudolf Usselmann
081-846-2302