JimBrown/MarlinTarantula #9: E steps

stylesuxx, 2017-08-10:
Hey,
in the config file there are two different values for E1_STEPS and E2_STEPS, both are labeled with "Stock extruder. If you have a Tevo Titan, try 376.14." in the comments.

Maybe E2 should be adjusted to be the same value.

JimBrown, 2017-08-10:
The configuration.h file in the main Marlin folder is for my specific printer. Please copy the configuration files from the example_configurations/Tevo Tarantula folder into the main Marlin folder.

In that file, you will find both E1 and E2 steps are set to 100, which is the starting point for the stock extruder. One should then do a calibration (see link in the comments). Also, yesterday I updated the comment to include information about the Titan extruder.

Please download the latest version and have a peek. Let me know if you see anything else.

Thanks for your input!

Regards,
...jim

stylesuxx, 2017-08-10:
Oh, alright. I assumed they were the same,... :-/
Sorry for bothering you and thank you for the great work (have been using the 01 config from the 1.1.x branch for the last couple of days)!

closed 2017-08-10 by stylesuxx