JimBrown/MarlinTarantula #15: Error compiling Arduino - marlin tarantula easy config 1.1.x

f-wilhelm, 2017-10-25:
Hi Jim

I have Problems with compiling the Marlin.ino (Tevo Tarantula). I have made my changes in the configuration.h and when I try to compile with Arduino IDE, I got this error Messages:
------------------------------------------------------------
### Compile Error

See attached pdf-file, please.

Nothing happend (no update of firmware). Can you please help me? What have I done wrong or what have i to do now?

regards
Friedrich
[Fehlermeldungen.pdf](https://github.com/JimBrown/MarlinTarantula/files/1415551/Fehlermeldungen.pdf)

JimBrown, 2017-10-25:
Please attach your configuration.h and configuration_adv.h files (in a .zip.) Thanks.

f-wilhelm, 2017-10-26:
Hi Jim 

Here they are, thank you.
[Archive.zip](https://github.com/JimBrown/MarlinTarantula/files/1418299/Archive.zip)

JimBrown, 2017-10-27:
You have enabled the SN04 sensor, but you have not disabled MANUAL bed leveling. You need to do that and then enable one of the other bed leveling types.

f-wilhelm, 2017-10-27:
Thank you, I will try it next week, I have a few days off and my Tevo is in my office.

Another question: in your wiki you describe with "how to center your prints" and "how to determine your z-probe offset" exactly the same steps. Does it mean, that I will have the same values in the define nozzle and define sensor (see my configuration.h)?

Regards
Friedrich

JimBrown, 2017-10-31:
As far as I can see, the "How to center your prints (EasyConfig)" and "How to determine your Z-Probe offset (EasyConfig) have different instructions, mostly. You should not have the same values for the nozzle and sensor settings, they should be different.

f-wilhelm, 2017-11-02:
Hi Jim 

The problem with compiling is solved, thank you very much for your helping. For me, this ticket can be closed.

I will have a closer look to the instructions offset and center the next days.
As I tried to do the bed levelling procedure, the nozzle crashed the heatbed, because the SN04 was over the edge at the right corner. Maybe it is my fault, I remember, that I had values for center and z-offset, I measured with the original firmware, I forgot to set them to zero before flashing the easy config 1.1.x.

I will flash again with the zero-values next week, if I have still this problem, I will open a new request.

Kind regards and thank you for your fabulous work
Friedrich

closed 2017-11-02 by f-wilhelm