JimBrown/MarlinTarantula #17: error in the

vatecq, 2017-10-30:
Hello,

I've just got tevo tarantula (mks v1.4) and there's no option of "store memory" in my LCD menu. I followed your instructions in your video and I got error message as below. So what could be the problem? I cannot script anything so I found you and I think you are successfull at your job but your instructions didnt work for me:/ what should I do now?

Arduino:1.8.5 (Windows 10), Kart:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\Marlin_main.cpp:267:0:

sketch\mesh_bed_leveling.h: In static member function 'static int8_t mesh_bed_leveling::cell_index_x(const float&)':

mesh_bed_leveling.h:77: error: 'MESH_MIN_X' was not declared in this scope

       int8_t cx = (x - (MESH_MIN_X)) * (1.0 / (MESH_X_DIST));

                         ^

In file included from sketch\Marlin_main.cpp:267:0:

mesh_bed_leveling.h:42: error: 'MESH_MAX_X' was not declared in this scope

   #define MESH_X_DIST ((MESH_MAX_X - (MESH_MIN_X)) / (GRID_MAX_POINTS_X - 1))

                         ^

sketch\mesh_bed_leveling.h:77:48: note: in expansion of macro 'MESH_X_DIST'

       int8_t cx = (x - (MESH_MIN_X)) * (1.0 / (MESH_X_DIST));

                                                ^

In file included from sketch\Marlin_main.cpp:267:0:

sketch\mesh_bed_leveling.h: In static member function 'static int8_t mesh_bed_leveling::cell_index_y(const float&)':

mesh_bed_leveling.h:82: error: 'MESH_MIN_Y' was not declared in this scope

       int8_t cy = (y - (MESH_MIN_Y)) * (1.0 / (MESH_Y_DIST));

                         ^

In file included from sketch\Marlin_main.cpp:267:0:

mesh_bed_leveling.h:43: error: 'MESH_MAX_Y' was not declared in this scope

   #define MESH_Y_DIST ((MESH_MAX_Y - (MESH_MIN_Y)) / (GRID_MAX_POINTS_Y - 1))

                         ^

sketch\mesh_bed_leveling.h:82:48: note: in expansion of macro 'MESH_Y_DIST'

       int8_t cy = (y - (MESH_MIN_Y)) * (1.0 / (MESH_Y_DIST));

                                                ^

In file included from sketch\Marlin_main.cpp:267:0:

sketch\mesh_bed_leveling.h: In static member function 'static int8_t mesh_bed_leveling::probe_index_x(const float&)':

mesh_bed_leveling.h:87: error: 'MESH_MIN_X' was not declared in this scope

       int8_t px = (x - (MESH_MIN_X) + 0.5 * (MESH_X_DIST)) * (1.0 / (MESH_X_DIST));

                         ^

In file included from sketch\Marlin_main.cpp:267:0:

mesh_bed_leveling.h:42: error: 'MESH_MAX_X' was not declared in this scope

   #define MESH_X_DIST ((MESH_MAX_X - (MESH_MIN_X)) / (GRID_MAX_POINTS_X - 1))

                         ^

sketch\mesh_bed_leveling.h:87:46: note: in expansion of macro 'MESH_X_DIST'

       int8_t px = (x - (MESH_MIN_X) + 0.5 * (MESH_X_DIST)) * (1.0 / (MESH_X_DIST));

                                              ^

In file included from sketch\Marlin_main.cpp:267:0:

sketch\mesh_bed_leveling.h: In static member function 'static int8_t mesh_bed_leveling::probe_index_y(const float&)':

mesh_bed_leveling.h:92: error: 'MESH_MIN_Y' was not declared in this scope

       int8_t py = (y - (MESH_MIN_Y) + 0.5 * (MESH_Y_DIST)) * (1.0 / (MESH_Y_DIST));

                         ^

In file included from sketch\Marlin_main.cpp:267:0:

mesh_bed_leveling.h:43: error: 'MESH_MAX_Y' was not declared in this scope

   #define MESH_Y_DIST ((MESH_MAX_Y - (MESH_MIN_Y)) / (GRID_MAX_POINTS_Y - 1))

                         ^

sketch\mesh_bed_leveling.h:92:46: note: in expansion of macro 'MESH_Y_DIST'

       int8_t py = (y - (MESH_MIN_Y) + 0.5 * (MESH_Y_DIST)) * (1.0 / (MESH_Y_DIST));

                                              ^

exit status 1
'MESH_MIN_X' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

JimBrown, 2017-10-31:
Hi, please zip up your configuration.h and configuration_adv.h files and attach them here.

vatecq, 2017-10-31:
Thank you so much Jim Brown, now it's totally working. Your configs are kind of a magic :)

closed 2017-11-02 by JimBrown