JimBrown/MarlinTarantula #18: Mesh Bed Levelling not working when printing

chanddan, 2017-11-01:
I have a Tevo Tarantula  and i'm trying to get the Bed Level (Mesh/Manual) feature to work.

I go through the steps in Prepare -> Bed Levelling -> Bed Level, everything seems fine. I save the settings.

But when I print the z-axis does not move. I have tried toggling "Bed Levelling On" and "Bed Levelling Off". Same issue.

I am doing something wrong or have I discovered a bug?

kingcocomango, 2017-11-01:
I have UML set, and while the options show up they literally do nothing. There is no response, on/off, attempting to start leveling does nothing, etc.

JimBrown, 2017-11-02:
@chanddan, you need to add something to your start script to load and activate the mesh you generated and saved. I believe it is M420 S1 to do that. Should be placed immediately after G28 in the start script. You can also add a Z parameter to that, which will slowly fade out the mesh correction up to the given height and then not use it above that height. For example, M420 S1 Z5 will load and activate the mesh compensation, and fade it out over the first 5 mm.

JimBrown, 2017-11-02:
@kingcocomango, I'm not sure what you mean my "UML"?

chanddan, 2017-11-02:
Thanks @JimBrown, I'll try M420 S1 and will report back soon.

chanddan, 2017-11-03:
@JimBrown M420 S1 did the trick, thanks again.

closed 2017-11-04 by chanddan