JimBrown/MarlinTarantula #20: Extruder motor doesn't work while printing

Sim0on, 2017-11-05:
Hello, thank you very much for all the stuff you did for the tarantula, 
I've just installed your firmware, everything goes right when I use the lcd command to move all the axis, including the extruder, but when I start a print from the SD card, everything works instead off the extruder motor stepper !! 
 I've only modified the extruder rotation sens in the "Configuration.h" and nothing else.
I use Simplify3D for GCODE.
Is there a wrong command created by simplify3D which block the extruder while printing ?

Thanks in advance (sorry for the mistake, I'm a native french^^)

JimBrown, 2017-11-06:
Sorry, I can't think of what the problem might be. Best bet would be to ask on the forums or the Facebook group. Good luck!

h3ct0r, 2017-11-06:
Maybe your hotend is not hot? There is a safe mechanism to prevent extruder motor for extruding filament if the hotend is cold. If I remember correctly the hot end temperature must be over 170 to the extruder to activate.

mrstanl3y, 2017-11-07:
Same issue, tried hooking the motor up to a different access to make sure its still good.   It is.   Then tried reflashing update.  Still extruder motor doesn't work.   I then tried the other extruder port on the board.  Nothing again.    I'm doing all the testing with a 200c hot end and 60c heat bed.   Wont extrude at all.

JimBrown, 2017-11-07:
Took another look, and yes, there may be an issue with the firmware. I'm investigating, but the issue is not with the "EasyConfig" stuff. It's somewhere deep inside the code.

JimBrown, 2017-11-07:
I've submitted a bug report to the Marlin repository. In the meantime, to get around this issue, set your slicer to *not* zero the extruder. It is the G92 E0 command that is broken.

Sim0on, 2017-11-07:
Nope, there is a real problem of the interpretation of G92 command by the firmware, I think also. So i resolve the problem by installing one customized 1.1.4 Marlin firmware that works perfectly with many options !

JimBrown, 2017-11-07:
The bug has been acknowledged on the Marlin Repository, and a fix will be implemented shortly. Will update when it is available here.

shufloo, 2017-11-07:
I'm so glad I wasn't the only one. I flashed Marlin and after that my printer wouldnt extrude during a print. I need to print parts for a client so thought it had all gone down the drain. Thanks for the support.

Sim0on, 2017-11-07:
Very good, good job ! thank you very much ! But for our knowledge, can you tell us what was the problem exactly and what have you modified in the firmaware to fix it ? thanks

JimBrown, 2017-11-07:
I did not modify the firmware, the Marlin guys did. Several days ago, they updated the firmware to add something to do with CNC machines. When they did, they removed a couple of lines of code for the G92 function that they thought were no longer needed. Turns out that they were still needed for the E axis. Once I brought it to there attention, they fixed it.

JimBrown, 2017-11-07:
Here's the bug report I opened: https://github.com/MarlinFirmware/Marlin/issues/8307