JimBrown/MarlinTarantula #21: bed levelling -> nozzle crash

f-wilhelm, 2017-11-08:
Hi Jim

I have a problem with bed levelling. 

Using Tevo Tarantula with large bed and sensor SN04. When I try to do bed levelling, the sensor is over the edge an the right end (see attached *.jpg) and the nozzle crashes the bed. the maximum x-distance to go would be 185mm, then the sensor is still on the bed. 

Can I set the maximum movement anywhere or what is the fault in this case, please? Is this the value on "define BED_MARGIN"? If yes, which value do I have to give there for maximum 185mm?

regards
Friedrich

[Tevo.zip](https://github.com/JimBrown/MarlinTarantula/files/1452797/Tevo.zip)

JimBrown, 2017-11-08:
No, BED_MARGIN is not what you want. You need to set your NOZZLE_X/Y offsets, and your SENSOR_XXX offsets. See these two articles on how to do that: 

https://github.com/JimBrown/MarlinTarantula/wiki/How-to-center-your-prints-(EasyConfig)

https://github.com/JimBrown/MarlinTarantula/wiki/How-to-determine-your-Z-Probe-offset-(EasyConfig)

f-wilhelm, 2017-11-09:
Hi

I hope, it is ok to do the movement of the axis with the "move axis" control of the display? I do not know how to send G1 X100 Y140 with the computer. I'm using Cura. Or shall I do the movement with Pronterface?
I have got X100 and Y137, this means nozzle_X 0 and nozzle_Y 3? On auto home the nozzle is now exactly over the mark in middle of bed and the display shows X100 Y140. On manual home Y it shows Y 3.0

After compiling and upload this values, Y is preset to 3.0. Now I want to do how-to-determine-your-z-probe. When I go to home Y, then it shows 3.0, do I have to send now 14times 10mm (140) with move axis Y, than it is on Position 143. Or does it mean, send it to position 140?

When I do "bed_levelling" now (sensor offset still 0,0,0,0, see question before), the nozzle crashes before it reaches the end position on the right side, because the sensor gets over the edge on the front and doesn't detect the bed.

JimBrown, 2017-11-09:
Pronterface does make it easier to move the nozzle around, but using the LCD will also work.

If you got 137 for the Y axis, then your Y endstop is not far enough back. Ideally, the endstop should be positioned so that when NOZZLE_Y is 0 (zero) and the nozzle is over the center of the bed, the Y should be at or more than 140. Otherwise, your bed will not be able to move all the way to 0 (zero). Try moving your endstop further back and start over with the first article.

Once you get that fixed up, then you need to follow the second article to set your sensor offsets properly so that it does not move the sensor off the bed when leveling.

f-wilhelm, 2017-11-14:
I still have problems to do that.

When I move the Y-endstop backwards to the end of the frame, I have to move the X- and Z-axis-carrier also back, because the nozzle is not over the bed in Y end position. I can use Y movement 274mm maximum now.
When I put the carrier in the middle of the Frame (Y direction), like shown at the mounting videos, I can't use the whole bed in Y direction.
I can't move the X endstop, it is not moveable and also the X-block touches the frame.

When I follow your instructions for "how to center your prints", I get the values X99 and Y139. This means, both nozzle values are +1 for uploading?
When I do this, after auto home the nozzle is exactly over the center mark and the info screen shows X100 Y140. When I put manually in X and Y home Position now, it shows X1.0 and Y1.0.

I don't know, how I should do the Z-Probe offset now. When I follow your instructions and move the nozzle to X100 and Y140, the nozzle is exactly over the mark. when I put manually 100 and 140 from home position, the nozzle is at X101 and Y141, because it has the offset 1 at home position.

What shall I do now or what have I to set for SENSOR_XXX, please?

JimBrown, 2017-11-14:
Did you mount the Y axis stepper using the acrylic pieces on both sides of the aluminum extrusion? If so, you can remove the acrylic piece that is on the other side from the stepper. This will allow you to move the Y endstop further back. And yes, you want to move the X/Z carrier back as well, but only as far as the nozzle is just at the front edge of the bed when the bed is all the way back and touching the endstop. Doing this should allow you to get the full 280 mm of bed travel.  I've attached a pic of what mine looks like.

[image]

Don't worry about the X endstop, leave it at 0 (zero). Being 1mm off at the center is not worth worrying about.

Once you get the above sorted out and the firmware uploaded, then you can work on getting the sensor offsets done. See the wiki article I linked to earlier for instructions. (I recently updated it to clarify it.)

f-wilhelm, 2017-11-16:
Hi

I mounted the Y axis stepper only with one acrylic Piece (in case of a Little accident during mounting...). The end stop is at the last possible endpoint of the frame. on the other side, the wheels are touching the acrylic pieces at a movement of 275mm. The only Thing, I can do, is to cut the acrylic pieces 10mm, than I have the total movement 280 with 5mm space.

[image]

[image]

But Y movement is not my priority problem now. I have still the problem, that the sensor doesn't detect anything in the right endposition of levelling process, the last point to detect the bed is at X 185. When it goes to 200, nozzle is still crashing, because the sensor has no bed to detect. "Detection point" of the sensor is 23mm right from the nozzle (measured with a metal ruler).
I don't know, how this should work when the sensor is not able to detect the bed?

JimBrown, 2017-11-16:
Did you set the SENSOR_xxx offsets as per this article? https://github.com/JimBrown/MarlinTarantula/wiki/How-to-determine-your-Z-Probe-offset-(EasyConfig)

f-wilhelm, 2017-11-17:
Mea culpa maxima.....

I always centered the nozzle, not the sensor. Don't know, why my brain was blocked, maybe because I was to desperate. 
Now it is working well, thank you very much.

I have taken some pics, maybe you want it for your instructions, if someone needs it for the green sensor. The green sensor (SN04) doesn't have a peak like yours. It is the BL-touch, I think? Is it better than the SN04?

Make additionally marks with 90 degrees:
[image]

a tape around the SN04, mark the middle in two directions:
[image]

center the sensor with the help of the marks:
[image]

regards and many thanks again
Friedrich

JimBrown, 2017-11-20:
Glad to hear you got it figured out! And thank you for the pictures with the SN04. I've updated the wiki article to use them, with credit to you.

closed 2017-11-20 by JimBrown