JimBrown/MarlinTarantula #23: a

odysseus2012, 2017-11-22:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

odysseus2012, 2017-11-22:
# Bug Report

## Description:
When homing the Axis the X axis limit switch is ignored. This behavor is only observed when running the homing script. If manually moving the X axis and manually activating the limit switch the firmware will stop the X movement. The Y axis and the Z axis operate normally.

## Expected behavior:
When running the homing script the X-Limit switch should trigger the Marlin Firmware to stop at the edge of the metal beam.

## Actual Behavior:
The X-carriage on the X-Axis hits the edge and the motor keeps trying to move it. A grinding noise occurs.

## Steps to Reproduce:
Menu -> Prepare -> Auto Home/Home X

_Zip file contains configuration and advanced configuration file._
[config-advConfig.zip](https://github.com/JimBrown/MarlinTarantula/files/1495718/config-advConfig.zip)

Thanks in Advance. :)

closed 2017-11-22 by odysseus2012

itsurboibrad, 2018-02-07:
Hi there, Im having the same problem with my printer. Im positive that its a software issue as the switch works when swapped for the y. It seems like theres no signal going through my x - port when it was working previously