JimBrown/MarlinTarantula #28: Compile error: fixed-point types not supported in C++

Trecoolius, 2017-12-10:
### Compile Error
When I compile I get an error:
```
In file included from sketch\MarlinConfig.h:30:0,

                 from sketch\configuration_store.h:26,

                 from sketch\configuration_store.cpp:185:

sketch\configuration_store.cpp: In static member function 'static void MarlinSettings::reset()':

Configuration.h:141: error: fixed-point types not supported in C++

 #define  hot_Kp 35.64R

                 ^

sketch\Configuration.h:576:23: note: in expansion of macro 'hot_Kp'

   #define  DEFAULT_Kp hot_Kp

                       ^

sketch\configuration_store.cpp:1429:26: note: in expansion of macro 'DEFAULT_Kp'

       PID_PARAM(Kp, e) = DEFAULT_Kp;

                          ^

In file included from sketch\configuration_store.cpp:193:0:

Configuration.h:142: error: fixed-point types not supported in C++

 #define  hot_Ki 3.72R

                 ^

sketch\temperature.h:104:30: note: in definition of macro 'scalePID_i'

   #define scalePID_i(i)   ( (i) * PID_dT )

                              ^

sketch\Configuration.h:577:23: note: in expansion of macro 'hot_Ki'

   #define  DEFAULT_Ki hot_Ki

                       ^

sketch\configuration_store.cpp:1430:37: note: in expansion of macro 'DEFAULT_Ki'

       PID_PARAM(Ki, e) = scalePID_i(DEFAULT_Ki);

                                     ^

exit status 1
fixed-point types not supported in C++

```
[Marlin.zip](https://github.com/JimBrown/MarlinTarantula/files/1545826/Marlin.zip)

Trecoolius, 2017-12-10:
EDIT: Sorry I was srunk and wasn't seeing an extra unwanted character in the E0 PID declarations.

closed 2017-12-10 by Trecoolius

JimBrown, 2017-12-11:
I know the feeling. Was a bit out of it last night myself. ;-) Glad you found the issue.

Have fun!