JimBrown/MarlinTarantula #32: Homing Order

Skybound420, 2018-01-04:
I don't know if this is a bug relative to Easy Config or the parent, but often when adjusting a Z offset, I may find the need to home and the printer moves X or Y first.  This should be Z up 5mm before any other movement is performed as damage may occur.  So far I've been lucky that my BLTouch incurred no damage when side crashing bulldog clips, but maybe not everyone else is.  Can you please alter the homing function to first lift Z out of harms way prior to X or Y movements?  thank you.

JimBrown, 2018-01-04:
There is a configuration item to ensure that Z is at least X mm above the bed. Currently it is set to 5 mm. This means that if the Z is below 5 mm, it will raise it to 5 mm then home X and Y, followed by Z. If the printer has been homed, and Z is above 5 mm, it will not raise Z. If the printer has not yet been homed, it will raise Z by 5 mm no matter what.

I have moved the setting for the Z homing height to the EasyConfig section, look for `#define Z_HOMING_HEIGHT` and adjust as needed.

There are also two settings to control the Z height when doing a G29 bed leveling, the first is the probe deploy height and the second is the height when moving between probe points. To simplify things, I have configured these two settings to use the same value as the Z homing height.

You'll need to download a new copy the firmware to use these changes.

Skybound420, 2018-01-04:
TBH, this may just be a conflict between Repetier Host (app I use for sending G codes) and Marlin, but I've found when below 5, usually at or very near true 0, if I were to send a G28, it would move X and/or Y first and if anything stand in the way, there'd be an NFL type of conflict, lol.  Thanks for the code adjustment, I'll install the new cut.  Also, since I have your ear, can you add acceleration and jerk variables into Easy Config?  I'm new to tweaking those, but in the groups, both seem a fairly popular topic of debate.

JimBrown, 2018-01-04:
Are you sending G28 from the command line? Or using some "buttons" in the GUI? If sending on the command line, G28 is G28 and there should be no difference what host you are using. If using the GUI and clicking a button, I have no idea what the host is sending.

Skybound420, 2018-01-04:
As it relates to this issue, I can't remember, but I've done both.  I'll have to try and reproduce the situation to see.

Gabrielzar, 2018-01-13:
Hi ! I have upgraded but I get a new error message when verifying ! :Arduino:1.8.3 (Windows 10), Tarjeta:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\Marlin_main.cpp: In function 'bool pause_print(const float&, const float&, const float&, const float&, const float&, int8_t, bool)':

Marlin_main.cpp:6499: error: 'PAUSE_PARK_Z_FEEDRATE' was not declared in this scope

       do_blocking_move_to_z(current_position[Z_AXIS] + z_lift, PAUSE_PARK_Z_FEEDRATE);

                                                                ^

Marlin_main.cpp:6502: error: 'PAUSE_PARK_XY_FEEDRATE' was not declared in this scope

     do_blocking_move_to_xy(x_pos, y_pos, PAUSE_PARK_XY_FEEDRATE);

                                          ^

sketch\Marlin_main.cpp: In function 'void resume_print(const float&, const float&, int8_t)':

Marlin_main.cpp:6679: error: 'PAUSE_PARK_XY_FEEDRATE' was not declared in this scope

     do_blocking_move_to_xy(resume_position[X_AXIS], resume_position[Y_AXIS], PAUSE_PARK_XY_FEEDRATE);

                                                                              ^

Marlin_main.cpp:6680: error: 'PAUSE_PARK_Z_FEEDRATE' was not declared in this scope

     do_blocking_move_to_z(resume_position[Z_AXIS], PAUSE_PARK_Z_FEEDRATE);

                                                    ^

exit status 1
'PAUSE_PARK_Z_FEEDRATE' was not declared in this scope

Este reporte podría tener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

JimBrown, 2018-01-14:
Did you copy the configuration files from the example configurations folder first? And then make the appropriate changes to the EasyConfig section?

I just tried the latest code and it verifies and uploads with no issues.

Gabrielzar, 2018-01-15:
Hi again ! I think I did but I will recheck again. I just want to mention that I have the SN04 sensor that came with my Tarantula and this sensor refuses to detect the black tarantula hotbed ! I have retested the sensor using aluminium foil ( kitchen foil ) and I works fine but obviously this film is too thin for the purpose ! I really do not know where to find I thicker film that would solve the problem could you give me any hint so that I can find some in my country ( Spain ) ? Maybe  you can make to me other material suggestions .... Many thanks in advance.

Gabrielzar, 2018-01-15:
Hi ! I have solved the problem I had with EasyConfig Upgrade by fully installing Easyconfig again and modifying the two config files under Tevo tarantula Examples. I beleive I had missed before reinstalling the latest Easyconfig and just changing the two config files ! Regards and sorry for the mistake.

snetel, 2018-01-17:
Hey, Great work so far but my Printer just doesnt Print anymore now 
The Printer Starts homing But the z axis goes  2 times up and the Printer resets (cools down and dont print) i copied the configuration.h filesi to the Main Ordner and dient overwrite ans Settings.
Do you have any idea?

Skybound420, 2018-01-18:
I believe the scope of this message board is to report bugs in the Easy Config fork of Marlin Firmware.  To seek help for any other issue at all, please post them in the Facebook group and many of the 40,000 members will be most helpful.

[https://www.facebook.com/groups/TEVO.3dprinter.owners/?ref=bookmarks](url)

betterwork1, 2018-01-19:
I am using 1.0.6  for my TeVo tarantula  I am able to communicate with the motherboard by using our arduino  examples and have made the board blink  but when I run Jim Brown's easy configuration  I get in error   It will not verify  and the Error says  Error compiling  and will not upload 
 I do use configurations from the Tevo tarantula please help

Gabrielzar, 2018-01-21:
Hi Jim ! I keep having problems with my Z- axis calibration. I have determined that my Z offset is -0.71mm  and stored it so that after homing the axis I get Z=5.71 on my LCD but whenever I start printing the printer moves up to Z=15 and starts printing in the air !  I am using the latest Easyconfig on my Tarantula with Large bed and SNO4 ( Green sensor )and I am printig directly from SD Card. Obviously I am missing something ... Could it be that my G-Code Starting script is wrong and it moves the print up ?. Thanks for your time and Regards.

JimBrown, 2018-01-21:
@snetel, @betterwork1, @Gabrielzar:

If any of you are still experiencing issues, please open a new issue.

Thanks.

closed 2018-01-21 by JimBrown