JimBrown/MarlinTarantula #34: The printer says "Heating E1" while it should be saying "E0"

magkopian, 2018-01-14:
## Bug Report
- Description: I have noticed that my printer says `Heating E1...` while it actually heats `E0`. I'm only using a single extruder and everything else appears to be working fine, the only thing that appears to be wrong is the message on the LCD. 
- Expected behaviour: The LCD should say `Heating E0...` while heating the first nozzle.
- Actual behaviour: Instead, the LCD says `Heating E1...`.
- Steps to reproduce:
  - Start a print
  - Wait for the bed to finish heating
  - Observe the LCD saying `Heating E1...` while it actually heats up `E0`.

Any changes I myself made to the configuration can be found [here](https://github.com/magkopian/MarlinTarantula/blob/EasyConfig-1.1.x/Marlin/Configuration.h), however I haven't changed anything related to the extruders order or their number.

JimBrown, 2018-01-14:
That is not a configuration item that can be changed (which is what my fork is about). It is in the actual code for the firmware. As such, it would be best to open an issue on the Marlin github site here: https://github.com/MarlinFirmware/Marlin/issues. However, if I recall correctly, that is actually the way it is designed. Heater E1 on the LCD refers to the first heater, connected to the E0 port. Yes, it is confusing.

closed 2018-01-14 by JimBrown

magkopian, 2018-01-14:
I generally agree, but I guess some controller boards call the connector for the first extruder `E1`, in which case `E1` indicating the first extruder would actually make sense. And since Marlin tries to be compatible with as many boards as possible, I guess `E1` would make a good default.

So, I think this is not really an upstream related issue, however if the goal of this fork is to not touch the sources at all and limit the scope only to editing the config files, I can understand that.