JimBrown/MarlinTarantula #35: [HELP] EasyConfig with SSR-80DD (40,50, etc releays)

ergoz, 2018-01-23:
Hello Jim! First of all big thanks for your work. I need help with EasyConfig. 
I want to use Tevo Tarantula with SSR-80DD from China, and to prevent it overheat with fast on/off states i want to use hysteresis for bed heating, but i dont understand how to correctly setup it in config. I think problem in my English :) Can anybody tell me in simple english how to configure firmware to decrease switching speed of releay? Or make page on Wiki.
Best regards, and sorry for my bad English

roygilby, 2018-01-23:
It is my understanding that a Solid State Relay is not suitable for use with PID control.  I think you should have a look in your copy of Marlin, in the Configuration_adv.h file, and configure "Bang-Bang" control for the heated bed.  Bang-Bang style control of the Heated Bed should work OK with the Solid State Relay. You'll also need to go into configuration.h and disable PID control for the Heated Bed.

Unfortunately, a search on Youtube has not revealed any videos that discuss using the "Bang-Bang" temperature control in Marlin FW.    A Google search doesn't reveal much more.  

There are some references in the Marlin Firmware Documentation Website that discuss enabling "Bang-Bang" heated bed control, including "BED_HYSTERESIS", but so far I haven't found any details on choosing a value for BED_HYSTERESIS.    The following is the link to the Marlin Firmware Documentation web site.

I hope that helps,

RoyG

http://marlinfw.org/docs/configuration/configuration.html

JimBrown, 2018-01-24:
This is not really an EasyConfig issue. Best bet would be to ask in the Marlin Repository issues.

That being said, my suggestion would be to disable `#define PIDTEMPBED` to turn of PID control of the bed and force it to `bangbang` mode. Enable `#define BED_LIMIT_SWITCHING` to use bed hysteresis and edit configuration_adv.h, `#define BED_HYSTERESIS 2` to slow the relay switching down. Bed hysteresis defines the temperature variation that will turn the relay on and off, in degrees C. I believe that (with the default of 2) that if the bed temp is set to, say, 90, that it will turn on the heater at 89 and turn it off at 91.

closed 2018-01-24 by JimBrown