JimBrown/MarlinTarantula #37: Full graphic sidplay

f-wilhelm, 2018-01-25:
Hi Jim

I tried to take a full graphic display (12864). The display is only highlighted, but no characters are seen. Turning the 10k-pot has no effect.

When I made changes for grphic display in configuration.h, I got error messages during compiling from the Arduino IDE.

Arduino: 1.8.5 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

"D:\Daten\3D\MarlinTarantula-EasyConfig-1.1.x_180118\MarlinTarantula-EasyConfig-1.1.x\Marlin\Marlin.ino:45:110: fatal error: U8glib.h: No such file or directory

     #include <U8glib.h> // library for graphics LCD by Oli Kraus (https://github.com/olikraus/U8glib_Arduino)
                                                                                                    ^
compilation terminated."

I put the U8glib_Arduino into the lib folder of my Arduino install, but nothing happens (after a new start of Arduino, of course).

[Marlin.zip](https://github.com/JimBrown/MarlinTarantula/files/1664304/Marlin.zip)

regards
Friedrich

JimBrown, 2018-01-25:
Try using the Library Manager within Arduino IDE. In 1.9.0 Beta you will find it under "Tools", "Manage Libraries" (Can't remember exactly where it is in earlier versions). In the Library Manager dialog, type "u8glib" in the "Filter your search..." box. Scroll to the bottom and you should see u8glib. Install it and you should be good to go.

JimBrown, 2018-01-25:
Here's the guide on using libraries: https://www.arduino.cc/en/Guide/Libraries

closed 2018-02-06 by JimBrown