JimBrown/MarlinTarantula #38: M303 and E3D V6 Heating Failed on v2.0.x

DJ-666, 2018-01-27:
### Bug Report
- Description: M303 results in a Heating Failed error with the E3D V6 hotend
A. manually setting nozzle temp to 200 works perfectly, consistent and smooth increase to target
B. E3D V6 hotend is defined in configuration.h
C. ONLY fails on version 2.0.x, works fine on 1.1.x

- Expected behaviour: M303 PID Auto-Tune completes successfully

- Actual behaviour: M303 starts, nozzle reaches approx 100-130 degrees then heating stalls, loses temp and then results in the Heating Failed error.

- Steps to reproduce:
  - With an E3D V6 hotend
  - send Gcode M303 E0 C8 S200 to printer
  - Wait for failure

*Cannot upload due to being on mobile*

JimBrown, 2018-01-27:
Not really an EasyConfig issue. May I suggest that you post this issue on the MarlinFirmware issue tracker. But, before doing that, try downloading a new copy of the firmware and giving it a try. I'm pretty sure that they have been doing things to the temperature routines over the last week or so.

Also, I am in the process of upgrading my printer to using a genuine E3D V6 (with direct drive Titan) so will be able to see if I have the same issue in the near future. May not happen for a day or three though.

DJ-666, 2018-01-27:
Thanks, I will raise the issue there, the current version was only downloaded a couple of days ago and from reading the commits over at the Marlin git it appears nothing related to the issue has been committed since then unless it was bundled in with other fixes

JimBrown, 2018-01-27:
Yeah, definitely something going on with 2.0.x... Fired my printer up this morning (with the stock hotend) and it would not accept temp commands. Just ignored them. Weird. Uploaded 1.1.x and it works fine.

Life on the bleeding edge...

DJ-666, 2018-01-27:
Scrap that, I just checked and I didn't actually download the file when I got the printer running again I downloaded it on 21/01 which was a couple of days before a issue with M303 Heating Failure was closed so it may be fixed!

Hahaha it's the sacrifice you make to get the latest features!

closed 2018-02-06 by JimBrown