JimBrown/MarlinTarantula #39: extruder selection

busrevi, 2018-02-05:
hi brown,
I set up your marlin tarantula update. If only one extruder is selected, E0 is activated by default. How can we do E1 as a prospect in this software.
Thankyou

JimBrown, 2018-02-06:
Edit "pins_RAMPS.h" file. Find these lines:

JimBrown, 2018-02-06:
Edit pins_RAMPS.h file. Find these lines:
```
#define E0_STEP_PIN        26
#define E0_DIR_PIN         28
#define E0_ENABLE_PIN      24
#ifndef E0_CS_PIN
  #define E0_CS_PIN        42
#endif

#define E1_STEP_PIN        36
#define E1_DIR_PIN         34
#define E1_ENABLE_PIN      30
#ifndef E1_CS_PIN
  #define E1_CS_PIN        44
#endif
```
Change `E0` to `E1` and change `E1` to `E0`. Like this:
```
#define E1_STEP_PIN        26
#define E1_DIR_PIN         28
#define E1_ENABLE_PIN      24
#ifndef E1_CS_PIN
  #define E1_CS_PIN        42
#endif

#define E0_STEP_PIN        36
#define E0_DIR_PIN         34
#define E0_ENABLE_PIN      30
#ifndef E0_CS_PIN
  #define E0_CS_PIN        44
#endif
```
Save the file and reflash the firmware.

This is not really an EasyConfig issue. If you still need assistance, best bet would be to ask on the Facebook group (https://www.facebook.com/groups/TEVO.3dprinter.owners/), or on the official Marlin repository (https://github.com/MarlinFirmware/Marlin/issues).

closed 2018-02-06 by JimBrown

p4ulv, 2018-11-12:
Thank you so much !