JimBrown/MarlinTarantula #41: Tevo tarantula easy_config printer halted please reset

Autiwa, 2018-02-19:
### Bug Report
- Description: After flashing with the latest easy_config 1.1 software, I want to prepare for PLA heating. can't print anymore. The bed doesn't heat, and at some point I get the error "printer halted please reset" (while the bed is almost cold and the temperature on screen doesn't change)

- Expected behaviour: temperature for bed and hotend reach their setting
- Actual behaviour: Hotend heats fine, but bed stays cold
- Steps to reproduce:
  - Flash tevo tarantula with latest easy_config 1.1 software, then try to heat the bed

[Marlin.zip](https://github.com/JimBrown/MarlinTarantula/files/1736400/Marlin.zip)

Please note that:
* this was working fine with the default software of my printer. 
* I received my printer less than a month ago
* I have a 220*220 bed
* a single extruder (I changed nothing in the printer, except adding a glass bed that worked fine with the default firmware
* I'm new to 3D printing so I might miss something obvious (that's why I waited a bit, to search on the web before)

While I'm not sure if it's a bug or not, This version of the software should work well with Tevo Tarantula since it's the default one. I couldn't find Configuration.h and Configuration_adv.h for tarantula in the "example_configuration" anymore. 

I know at least another person in the same case (he had Marlin before, but when trying the easy_config version, his tevo tarantula doesn't work anymore)

JimBrown, 2018-02-19:
Your configuration files are NOT from my EasyConfig fork. Please download the latest zip from here: https://github.com/JimBrown/MarlinTarantula (click "Clone or Download", then click "Download Zip") and try again. Please be sure to read the ReadMe file, and the comments at the top of the configuration.h file.

closed 2018-02-19 by JimBrown

Autiwa, 2018-02-20:
That's exactly what I did. I downloaded the .zip file from the branch EasyConfig-1.1.x. 

But I'm starting to realize that as unlucky as I am, something might have happened during the flash. Because no matter what I try, the bed won't heat, with any other firmware I try. I no longer think the firmware is the problem, I think it's hardware, even if I don't know what.

JimBrown, 2018-02-20:
While that is possible, please try again but use the configuration files that are included in the Zip. The ones you attached to your original post were not from the Zip. I suspect that you copied them from your old firmware. Use the ones included in the zip, and read through the comments included in the EasyConfig section to figure out what needs to be enabled/disabled and/or changed. If your bed still refuses to heat up then yes, I would suspect a hardware problem.