JimBrown/MarlinTarantula #42: #define UBL

CSTE101, 2018-03-26:
Hi!
When I uncomment #define UBL I got error:

SanityCheck.h:808: error: #error "AUTO_BED_LEVELING_UBL (<=1.1.8) always has RESTORE_LEVELING_AFTER_G28 enabled. To keep this behavior, #define RESTORE_LEVELING_AFTER_G28. To keep it disabled comment out this line in SanityCheck.h."

When I uncomment #define #RESTORE_LEVELING_AFTER_G28 I got error:

Marlin_main.cpp:11566: error: 'class Planner' has no member named 'apply_leveling'

                 planner.apply_leveling(current_position[X_AXIS], current_position[Y_AXIS], z1);

JimBrown, 2018-03-27:
Working fine for me. Please download a new copy of the firmware and try again. Note: Do not copy your current configuration.h and configuration_adv.h files to the new copy. Edit the new copy to match your old copy.

CSTE101, 2018-03-27:
Hi, i downloaded fresh copy, uncomment: 
#define SN04  
and
#define UBL

I still have the same errors. Arduino IDE 1.8.5.

JimBrown, 2018-03-28:
Found the problem. And it is now fixed. You can download a new copy of the firmware, or you can edit configuration.h and enable this line:

CSTE101, 2018-03-28:
Stil same errors :(
When I commit: #define RESTORE_LEVELING_AFTER_G28

/var/folders/sb/sg0ny6m52313dcz4qnjjq9tw0000gn/T/arduino_build_642191/sketch/Marlin_main.cpp: In function 'void tool_change(uint8_t, float, bool)':
Marlin_main.cpp:11566: error: 'class Planner' has no member named 'apply_leveling'
                 planner.apply_leveling(current_position[X_AXIS], current_position[Y_AXIS], z1);
                         ^
Marlin_main.cpp:11567: error: 'class Planner' has no member named 'apply_leveling'
                 planner.apply_leveling(x2, y2, z2);
                         ^

JimBrown, 2018-03-28:
Appears to be a bug in the main Marlin code when using UBL with dual extruder/dual nozzle. I don't do the coding, just mess with the configuration, so you'll have to take this to the official Marlin repository issues page. In the mean time, you can get around it by using BILINEAR bed leveling, as that appears to still compile and work.

JimBrown, 2018-03-29:
I've opened an issue on the Marlin repository issues page. Will let you know when it is resolved.

CSTE101, 2018-03-29:
Thank you JimBrown! :)

JimBrown, 2018-03-30:
Patch for this issue has been implemented. Download a new copy and transfer your settings. Should be good to go now.

Thanks.

closed 2018-03-30 by JimBrown

CSTE101, 2018-03-30:
Thanx!