JimBrown/MarlinTarantula #45: Compile error in temperature.cpp

gjelsoe, 2018-04-24:
Getting this compile error on last branch of MarlinFirmware:1.1.x. code

```
Arduino: 1.8.5 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\temperature.cpp: In static member function 'static void Temperature::checkExtruderAutoFans()':

temperature.cpp:561: error: 'current_temperature_chamber' was not declared in this scope

     if (current_temperature_chamber > EXTRUDER_AUTO_FAN_TEMPERATURE)

         ^

exit status 1
'current_temperature_chamber' was not declared in this scope
```

when changing this from
`#define E0_AUTO_FAN_PIN -1`
to 
`#define E0_AUTO_FAN_PIN FAN1_PIN`
 in `Configuration_adv.h` file.

JimBrown, 2018-04-24:
Sorry, I have not played with that stuff, and it really does not have anything to do with my EasyConfig changes. Recommend you ask this in the Marlin repository issues page here: https://github.com/MarlinFirmware/Marlin/issues

closed 2018-04-24 by JimBrown

gjelsoe, 2018-04-24:
Marlin repository doesn't seem to be updated with the same code.

JimBrown, 2018-04-25:
Did you switch to the bugfix-1.1.x branch? That is what EasyConfig-1.1.x is a fork of.

gjelsoe, 2018-04-25:
Sorry about that, forgot to select the Bugfix.....

But they have corrected the error now..