JimBrown/MarlinTarantula #49: Bltouch wont deploy when homing

pirogeth, 2018-07-28:
I have enabled the bltouch on the firmware. but the probe wont deploy. it works when it does the self test and deploys and stows when i do it manually. but it wont work while i do a home. Let me know what you need to see.

mhaken, 2018-07-28:
I have this problem too.

JimBrown, 2018-07-28:
Download a new copy of the firmware and try again. I noticed in this mornings update that there were some fixes to the BLTouch code.

pirogeth, 2018-07-29:
Are you able to include some code for the bltouch for the tevo flash. i tore it apart the other day and noticed that its not a conventional setup I have the original firmware if you want to take a look at what i mean. in the control box its wired up funny. not the way i would do it if i was installing a bltouch.  
https://www.dropbox.com/s/17s9du8untwg5np/Flash_TMC2100_BL.rar?dl=0

JimBrown, 2018-07-30:
Wish I could help, but all I really do is mess around with the configuration files. I don't do the code itself. You might want to try the official Marlin repository issues page.

closed 2018-08-22 by JimBrown

julianCast, 2018-09-05:
@pirogeth did you solve the problem updating? I'm facing the same issue,it works,, but sometimes, caused by some actions I'm not sure,like finish printing or a simpe reboot, the bltouch will crash into bed. I'm thinking about trying with marlin 2 jim brown.