JimBrown/MarlinTarantula #55: Code assumes print bed is 200mm wide, my OOB tevo tarantula bed is 220mm wide (x-axis)

nikolac, 2018-10-12:
I purchased the Large Bed Tevo Tarantula in August 2017. It's 220x280mm. After struggling with UBL and z-offsets, I realized that LARGE_BED setting only sets the Y to 280. I had to change the X_BED_SIZE to 220 in the non-easy section of Configuration.h. 

I have to imagine this is causing issues for other people too.

JimBrown, 2018-10-13:
Both the large bed and the standard bed are 220 on the X axis. The standard bed is also 220 on the Y axis. Not sure what the actual Y axis size is on the large bed. Pretty much all heat beds are designed that way, they don't really use the outside 10 mm. If you were having issues with the X axis set at 200, then most likely there was something not quite right with your nozzle and/or probe offsets. Did you follow the wiki articles I've posted here on centering your prints and setting your probe offsets? And did you set your slicer to use 200 on the X axis?

nikolac, 2018-10-13:
I followed both of the wiki articles and read through another issue that produced some of the SN04 images. I had everything aligned appropriately, but there was always a larger than expected area of missed bed on the far side of the X axis.

It's just unintuitive. Sure, you might not use the 10mm on the outside of the bed. But, when I have everything centered appropriately and there is a setting called BED_MARGIN and I set that to 10mm, and then I get a 20mm gap...it makes me thing I messed something up.

Wouldn't it make more sense to set the X_BED_SIZE, Y_BED_SIZE to their actual values, then default the margin to 10? Or X_MARGIN to 10?

closed 2018-12-20 by JimBrown