JimBrown/MarlinTarantula #60: Z_PROBE_OFFSET_FROM_EXTRUDER not honored

cdaringe, 2018-12-21:
# problem

- `Z_PROBE_OFFSET_FROM_EXTRUDER` has no effect on Z position, post homing

# Expected behavior

- on setting `Z_PROBE_OFFSET_FROM_EXTRUDER` to a large value, e.g. 10, going to `G1 Z0` should go 10mm below the actual trigger point.  I am using a SN04 sensor

# Actual behavior

- `G1 Z0` goes strictly to the trigger point, which is ~ `Z0`

#### Additional Information

- I am using a puck to level my bed.  the puck has flat foil attached to it, and triggers the sn04 pretty far away.  this is why I cannot set `Z_PROBE_OFFSET_FROM_EXTRUDER = 0`
- `Z_MIN_POS` is a large negative value. I don't think this should be relevant because _i'd assume_ post Z HOME, the current Z position would take into account `Z_PROBE_OFFSET_FROM_EXTRUDER`
- `Z_PROBE_LOW_POINT`, again, shouldn't be relevant, but set to a large negative value

my configuration file: https://github.com/cdaringe/MarlinTarantula/blob/469d77a5258c39ab493caa6bfd3613286c4f786c/Marlin/Configuration.h

my configuration file diffed against this repo's 2.x branch: https://github.com/JimBrown/MarlinTarantula/compare/EasyConfig-2.0.x...cdaringe:EasyConfig-2.0.x (nothin too wild!)

cdaringe, 2018-12-21:
errr Z_MIN_POS clearly doesn't do what I thought it did, but the issue still stands

JimBrown, 2019-02-11:
Apologies for the delay, I've been rather occupied with other stuff.

I have never used the Z_PROBE_OFFSET_FROM_EXTRUDER (I have a BLTouch, so am in the same situation as you). Instead, I adjust the offset in my start script as per this wiki article:

https://github.com/JimBrown/MarlinTarantula/wiki/Setting-the-Z-height-offset

Hope that helps.

closed 2019-02-11 by JimBrown