JimBrown/MarlinTarantula #63: MINTEMP getting thrown on E1 hotend

nrgbistro, 2019-02-10:
### Description

When using a fresh install of this firmware, MINTEMP is thrown every time on E1 no matter what settings I change. The thermistor seems fine, as the temperature readout reads room temp and increases for a short time while the hotend heats up until MINTEMP is thrown.
[Printer Error](https://drive.google.com/open?id=1tI_1Y_4nAgeW_kTQVRIVyW6O2Dl0lPIe)
### Steps to Reproduce

As far as I can tell, just using the firmware causes this bug. I don't have anything special setup on my board, pinouts, or anything else really. Just a stock single extruder Tevo.

**Expected behavior:** [What you expect to happen]

Printer heats up to selected temperature

**Actual behavior:** [What actually happens]

Printer heats up for about 3-5 seconds, to no more than about 30 degrees celsius (according the the thermistor), then throws MINTEMP on E1

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files:
No need; firmware is not modified at all, aside from enabling the SN04 touch sensor and defining tripoint leveling (Both via the 'easy config section)
* Provide pictures or links to videos that clearly demonstrate the issue:

JimBrown, 2019-02-11:
That is weird. As far as I know, a stock Tarantula with a single nozzle uses the E0 heater, not the E1 heater. Send me your configuration.h and configuration_adv.h files anyway and I'll make sure they don't have something weird going on.

nrgbistro, 2019-02-11:
[configuration.h](https://drive.google.com/open?id=1gYZtRjoGxX8R2zBjFtMXGrqaJlOzBkwj)
[configuration_adv.h](https://drive.google.com/open?id=14jrTPPttyq0frj2pvxFmWgirZDIph0py)

JimBrown, 2019-02-12:
Ok, I don't see anything in your configuration files that would cause this issue. Since my repository is just for an easy way to configure Marlin for the Tevo Tarantula and is not really meant to provide support for Marlin itself, I recommend that post your issue in the official Marlin repository here: https://github.com/MarlinFirmware/Marlin/issues.

closed 2019-02-12 by JimBrown