JimBrown/MarlinTarantula #64: Print From Sd Not Working

djice024, 2019-03-21:
### Description

I have Upgraded To EasyConfig 2.0.x, after upgrading my printer with an inductive auto leveling sensor and a genuine E3D v6 hot-end. The Problem is that after the upgrade printing from SD doesn't work anymore. The printer detects the SD card and I can click on a gcode file to print, but the print doesn't start. The only thing that happens is that it puts the filename at the bottom of the screen.

### Steps to Reproduce

1. Flash Marlin EasyConfig 2.0.x.
2. Slice Gcode and upload it to the SD card.
3. Insert it in the printer slot.
4. Navigate to print from SD.
5. Select the file.

**Expected behavior:** 
The print starting normally.

**Actual behavior:** [What actually happens]
The Filename is being listed at the bottom of the screen, but nothing else is happening. The hot-end nor the heated bed is warming up.

Attached are my config files:
[3D Printer.zip](https://github.com/JimBrown/MarlinTarantula/files/2993647/3D.Printer.zip)

JimBrown, 2019-03-22:
Your config files are from an old version of Marlin. Try downloading the latest version here and transfer your EasyConfig settings from your old configuration.h file to the new configuration.h file (do not copy the old file over the new file). Compile, upload and see if that fixes things.

You can get the latest version here: https://github.com/JimBrown/MarlinTarantula. Click the "Clone or Download" button, the click "Download ZIP".

djice024, 2019-04-02:
I upgraded the firmware, but that didn't seem to work either. After some time experimenting re-installing cura made it work again. Thanks for the help.

closed 2019-04-06 by JimBrown