JimBrown/MarlinTarantula #65: SD Card, Cannot read

3D-og, 2019-04-05:
Updated and enabled SD card support.

Display says SD card inserted but when I navigate to the SD card, says NO SD CARD.

Worked before update so not hardware issue.

Tried several SD cards including fresh formatted card with no files.

Tried slowing the SD speed to 1/2 and 1/8.

Still no luck.

Stock TEVO
Dual Print, Single nozzle.
Large bed.
Gen 1.4 board.

JimBrown, 2019-04-06:
Tested on my printer (original Tevo with MKS Base 1.4 board) and it is working as it should. Since the purpose of my repository is only to provide an easy way to configure the firmware and that none of the actual code is modified, I can only recommend that you post your issue in the official Marlin repository issues section at https://github.com/MarlinFirmware/Marlin/issues

Good luck and I hope you find and fix the issue.

closed 2019-04-06 by JimBrown

3D-og, 2019-04-06:
Thanks for the reply.

Stubbornly working on the problem and making some progress.

I will post my problem and any solution I find to the /issues

Thanks again for the reply.

From: Jim Brown
Sent: Saturday, April 6, 2019 10:18 AM
To: JimBrown/MarlinTarantula
Cc: 3D-og ; Author
Subject: Re: [JimBrown/MarlinTarantula] SD Card, Cannot read (#65)

Tested on my printer (original Tevo with MKS Base 1.4 board) and it is working as it should. Since the purpose of my repository is only to provide an easy way to configure the firmware and that none of the actual code is modified, I can only recommend that you post your issue in the official Marlin repository issues section at https://github.com/MarlinFirmware/Marlin/issues

Good luck and I hope you find and fix the issue.

tomukas1990, 2019-05-30:
Any success with this problem? I have same problem with same board (GEN 1.4) and Marlin 2.0 firmware...

3D-og, 2019-05-31:
yes.. I traced the problem back to ‘sort file names’ option.

I disabled that feature and works now.

From: tomukas1990
Sent: Thursday, May 30, 2019 2:55 PM
To: JimBrown/MarlinTarantula
Cc: 3D-og ; Author
Subject: Re: [JimBrown/MarlinTarantula] SD Card, Cannot read (#65)

Any success with this problem? I have same problem with same board (GEN 1.4) and Marlin 2.0 firmware...