JimBrown/MarlinTarantula #66: m49 error

tdude0317, 2019-04-14:
when i try to verify the code I get this error even when i haven't changed anything

sketch\src\gcode\bedlevel\ubl\M49.cpp:34:22: error: no 'void GcodeSuite::M49()' member function declared in class 'GcodeSuite'

 void GcodeSuite::M49() {

                      ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560

how would i fix this?

JimBrown, 2019-04-15:
It is compiling fine for me. Are you using the latest version of Arduino IDE?

tdude0317, 2019-04-15:
yes I am but it still comes up with the error
i will try uninstalling and reinstalling the files and possibly arduino though

tdude0317, 2019-04-15:
ok reinstalling both the code and arduino did it

closed 2019-04-15 by tdude0317