JimBrown/MarlinTarantula #72: Micro vibrationos printing on curves.

paulorfo, 2019-11-06:
Hi, I have a Tevo Tarantula, was using EasyConfig 2.0.X and manual bed leveling. so far so good.

I installed a BLTouch and noticed the printer was vibrating while printing on curves.
Flashed the Marlin BugFix 2.0.X using Arduino 1.8.10. Before that I optimized it to work on a Tevo Tarantula. Same thing happened, but it also slows down printing on straight lines, it vibrates and it ends up with some "columns".
Tried the Marlin BugFix 1.1.X version after optimize it as well... Works great. EasyConfig 1.1.x also works normally.
I used the same Gcode for all my tests.
I played with jerk and acceleration.
Tried to flash the firmware using older version of Arduino.
Also tried to remove the BLTouch and tested all different firmware. 1.1.x works fine, 2.0.x vibrates.
Inspected thoroughly the printer for broken brackets, lose screws, damaged belts, stepper motor current, etc.

Re-flashed EasyConfig 1.1.X and it's working perfectly.

Putting on short, both 1.1.x (BugFix and EasyConfig) work well and both 2.0.x vibrates.
I can't understand why this vibration appeared, because I've been using the EasyConfig 2.0.x for quite some time!

Apparently nothing has a major impact on the vibration.
I ran out of ideas and possible solutions.

Jerk X/Y = 7
Acceleration = 800

Any idea??
Thanks.

JimBrown, 2019-11-09:
There have been a number of changes made to the firmware regarding jerk and curves. Please download a new copy of the firmware and try again. Do not copy the old configuration file over the new ones. Edit the new ones to match what you have.

closed 2019-11-09 by JimBrown

paulorfo, 2019-11-10:
I did it. But now, at the beginning of a print, X and Y axis do the homing process normally.
Z axis doesn't respond and a message "STOPPED" is shown on the display.